mainsail-crew/mainsail #937: improvement mainsail 2.21

Ipermax2000, 2022-06-28:
### Requested feature:

it is useful to add the ability to protect system files, printer.cfg mainsil.cfg moonraker.conf klipperscree.conf... to prevent them from being deleted by mistake. the way they present themselves is similar to the history could happen by distraction to be able to delete them by mistake. for example put in the menu an item to be able to temporarily hide the files

lixxbox, 2022-06-30:
I am not sure that this is really helpful. A little self-responsibility and most importantly backups should be expected from every user, otherwise we will someday have security advisories and and pop-ups everywhere in Mainsail.

But that's just my humble opinion.

Ipermax2000, 2022-06-30:
I didn't know how to ask the question. I understand that things can get complicated. however, put yourself in the perspective of those who want to offer the printer to customers. for personal use it's fine you're right but if I wanted to propose the machine to a company or private people, I would like to be sure that they do not end up damaging the interface and compromising the system. you see the issue not only for us as end users but also as people who then want to propose the printer to third parties...

th33xitus, 2022-06-30:
How do you ensure that the user/company won't use the same function you used to hide the files to make them visible again?

th33xitus, 2022-07-01:
In my opinion, what he is asking for falls within the scope of a system administrator anyway. We can't protect system files with Mainsail. Yes, we could hide them, but as already alluded to, I don't see the point. A function that allows you to hide individual files simply does not protect them from being displayed again by anyone and then edited or even deleted.

Removing write access to printer.cfg or moonraker.conf might work to prevent their contents from being changed. You lose the automatic save function of Klipper (SAVE_CONFIG), but this might be acceptable for presentation purposes. However, this again does not protect the files from being deleted.

TL;DR: Ask a Linux system administrator you trust about how best to address this problem. Personally, I don't see it as Mainsail's job.

Ipermax2000, 2022-07-01:
I understand what you mean, it would be enough to protect the system wires from accidental deletions, put them in a way to make their damage more difficult. I think of something and then let you know. regardless of this, however, the interface is done very well you have done a great job.

Ipermax2000, 2022-07-01:
adv example it could be useful a button appropriately named (for example system file), inserted in the interface that when clicked hides all the system files and only deliberately clicking on it makes the files appear ... so at that point there must be the will to do damage and not an accidental situation I do not know if I explained myself then I make an image thank you very much for now

th33xitus, 2022-07-01:
> it could be useful a button appropriately named (for example system file), inserted in the interface that when clicked hides all the system files and only deliberately clicking on it makes the files appear ... so at that point there must be the will to do damage

i would guess that such a button would be useless for probably 98% of the users. the normal workflow is to go to the machine page when either updates need to be made or configuration files need to be adjusted. i can't imagine a regular workflow where i want to hide my configuration files so i don't accidentally delete them. it would also be too much effort for me to show the files every time before i can see them and edit them. i always switch explicitly to the machine page **when** i want to make changes to the files. for this reason you should always be careful which file you are about to edit or delete.

still accessible via the machine-page are the system-updates to which the same way of thinking should be applied. this would mean that you should not allow these updates for other users either. in the end it would mean that you have to block the whole machine-site for unauthorized persons, because actions can be executed over the whole site which can influence the system negatively. i didn't test it, but maybe this would even be possible via a corresponding NGINX configuration, in which the access to the machine-site is blocked directly by NGINX itself.

alternatively, you could simply make a hidden folder in your configuration folder. this folder contains your actual configuration files. in the parent directory you then have a "dummy printer.cfg" and "dummy moonraker.conf" whose content consists only of corresponding includes from the hidden directory. in the case of printer.cfg, however, this can also lead to problems in combination with SAVE_CONFIG. however, it would be a possible approach at least.

Ipermax2000, 2022-07-01:
I think I have not explained myself, it is not so tragic what I want to say

Ipermax2000, 2022-07-01:
I send an image so I make the concept well

Ipermax2000, 2022-07-01:
[example.webm](https://user-images.githubusercontent.com/25878762/176892841-910b9d02-233f-4bf2-98aa-2a264a606704.webm)

Ipermax2000, 2022-07-01:
example fast and safer practical system to safeguard files and not slow down their update

th33xitus, 2022-07-01:
Okay, i didn't expect your proposal to be **that** simple. 😄 I really didn't understand that your actual feature request is to make the file manager panel collapsible. That is actually something we can likely implement very easily.

Ipermax2000, 2022-07-01:
perfect. thank you very much, my English is bad but I make progress. we update hello and good job

PR #943 (2022-07-08): feat: allow collapsing of config file panel
fixes #937

some refactoring was also done to fix some weaker IDE warnings.

there is still one warning about a deprecated "slot" symbol:
https://github.com/th33xitus/mainsail/blob/eda51cb85d565f0a05898c8eeea42232fa88cd8b/src/components/panels/Machine/ConfigFilesPanel.vue#L117
i wasn't sure how to approach that though. @meteyou maybe you have an idea? would be nice if we get that fixed before merging as well.
closed 2022-07-08 by meteyou