mainsail-crew/mainsail #954: "Hide timelapse"  in console shows Timelapse: disabled, take frame ignored

daniel-volz, 2022-07-09:
### Mainsail Version:

2.2.1

### Browser:

Firefox

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

Setting the "Hide Timelapse" Setting in Console Filter, The Console still shows Messages like:
Timelapse: disabled, take frame ignored

### What did you expect to happen instead?

Setting the Filter should hide all Timelaps-Messages in Console

### How to reproduce this bug?

Set the Filter to "Hide Timelapse" and print a file containing TIMELAPSE_TAKE_FRAME before Layer Change

zellneralex, 2022-07-09:
This is not a bug.

The console filter filters the setup messages. The message you see can be enabled and disabled by Verbose GCode setting in the Timelapse setup.

we decided to split that as you in 99.99% of cases want to hide the setup request (They are used internally and will work)m but for debugging you might want to see the rest.

If you agree, then close it

daniel-volz, 2022-07-09:
Thanks for the fast respond. 
In that case, i will close the ticket

closed 2022-07-09 by daniel-volz