mainsail-crew/mainsail #956: Logitech c920 - Just One Camera Please

Xcion54, 2022-07-10:
### Mainsail Version:

2.2.1

### Browser:

Chrome

### Device:

Other (Please tell us below)

### Operating System:

Windows

### What happened?

How much money does someone need to make the Logitech C920 (Very Common Camera) work on Mainsail because the amount of Issues and glitches doesnt let me use this camera at all. thank you so much for all your work, please email me - jonny.williams35@gmail.com

### What did you expect to happen instead?

I wish we could connect at least this one model of camera to mainsail

### How to reproduce this bug?

It just does not work consistenlty I need a separate device to monitor the printer

### Additional information:

How much money does someone need to make the Logitech C920 (Very Common Camera) work on Mainsail because the amount of Issues and glitches doesnt let me use this camera at all. thank you so much for all your work, please email me - jonny.williams35@gmail.com

zellneralex, 2022-07-10:
Hi,

that’s not a mainsail issue, but anyhow I have Mainsail and a c920 working without issues for years now. I recommend to use crowsnest and as a starting point you can than use my config https://github.com/zellneralex/klipper_config/blob/master/crowsnest.conf. You will need to change the device name. If you run in trouble I recommend to ask at our discord in the webcam subforum

closed 2022-07-10 by meteyou

Xcion54, 2022-07-12:
> Hi,
>
> that’s not a mainsail issue, but anyhow I have Mainsail and a c920 working without issues for years now. I recommend to use crowsnest and as a starting point you can than use my config [...]

Oh ok I will check out Crowsnest thanks