mainsail-crew/mainsail #958: Toolhead control font size broken on Firefox

9R, 2022-07-10:
### Mainsail Version:

v2.2.1

### Browser:

Firefox

### Device:

Laptop

### Operating System:

Linux

### What happened?

When using Firefox the font on the toolhead control style 'circle' is not rendered as expected.
I have attached screenshots using Firefox and Chromium on my system for comparison.
[image]

### What did you expect to happen instead?

Font with correct size
[image]
.

### How to reproduce this bug?

Open the UI with firefox and activate toolhead control with style circle

### Additional information:

Firefox version: 101.0.1

th33xitus, 2022-07-10:
> Open the UI with firefox and activate toolhead control with style circle

Unable to reproduce. Although im using Firefox on Windows.

[image]

9R, 2022-07-11:
I have done some further testing.
The issue arises on a laptop with high resolution display (12.5" with 3000x2000). I have tested other devices with fullHD and 4K displays, that also run Linux and latest Firefox. There the font is rendered correctly. Strangely enough when I compare eck the font parameters with dev tools from FIrefox they are the same on the offending device and the one with the 4K display that renders toolhead control correctly.
I also tried using dev tools to modify font-size manually which was not possible, Below a certain threshold the font will not get any smaller, but I am no expert in webdesign so I'm not sure if i have exhausted all possibilities.
Non of the tested computers use any text-scaling or accessibility features that I know of. I have not noticed any similar rendering errors elsewhere in mainsail or on other websites with the laptop that exhibits the behavior.

I'm unsure how to proceed, since this seems to be isolated to a single device for now and I do not know how to reproduce the behavior on other machines and impact is mostly visual and there are probably more important issues to resolve.

closed 2022-08-23 by meteyou