mainsail-crew/mainsail #963: CSV export uses commas as thousands separator

Anarchid, 2022-07-13:
### Mainsail Version:

2.2.0

### Browser:

Firefox

### Device:

Laptop

### Operating System:

Linux

### What happened?

The CSV export option in printer history seems to use `,` as a thousands separator, which causes the numerical value to be broken into two columns, and the export invalid.

For example, a print that expended 22843 millimeters of filament will be exported as `22,843`. 

Doing some RTFS, it seems that the exporter is supposed to switch to using semicolons as column separator in such case; however, this also does not occur, and any attempt to import the produced CSV is doomed to parse `22,843` as two separate columns.

### What did you expect to happen instead?

CSV export should do one of these:
- never use thousands separator when formatting numbers
- always use semicolons as rows separator
- use semicolons as rows separator when the exported file would contain even one numerical value >= 1000.

### How to reproduce this bug?

Export history as CSV where any print exceeds 1000 mm of printed filament.

### Additional information:

I suspect that this is probably due to browser locale settings or some such nonsense.

meteyou, 2022-07-13:
pls upload a test expoert

Anarchid, 2022-07-13:
[print_history.csv](https://github.com/mainsail-crew/mainsail/files/9105484/print_history.csv)

Anarchid, 2022-07-13:
On some further thought, it looks like any numerical values are fed through toLocaleString [here](https://github.com/mainsail-crew/mainsail/blob/14fa54e369283bd1a206769fd38fd6e585dec9cb/src/components/panels/HistoryListPanel.vue#L1014), and then not checked for the presence of `escapeChar` like in case of `string` below.

meteyou, 2022-07-13:
i use "toLocalString" because CSV in some countries have ; to split columns. for example in german we use . for thousand separator instead of , and , for decimal point. thats why CSV files are different in some countries.

Anarchid, 2022-07-13:
Hmm, perhaps an addiitonal thousands separator check like [here](https://github.com/mainsail-crew/mainsail/blob/14fa54e369283bd1a206769fd38fd6e585dec9cb/src/components/panels/HistoryListPanel.vue#L943) with a switch to semicolon if `,` detected as thousands separator would be the best approach then.

closed 2022-08-02 by meteyou

drewmo, 2023-06-29:
This issue seems to have cropped up again when exporting print history. There is an unescaped comma in the `start_time` value.

You can see an example of this below:

[image]

And when viewing the exported csv in VScode:

[image]

The comma is not escaped in the `start_time` cell which is splitting the month / day value and year / time value into two columns.

meteyou, 2023-06-29:
@drewmo it's not the same issue. Only similar...

Which location / language do you use on your pc? Mainsail use the local time format from your pc, so I need this setting to reproduce the issue.

drewmo, 2023-06-29:
@meteyou I am on a Mac, using the following date/time settings:

[image]

Also using Firefox as my browser with English(US) as the language.

PR #967 (2022-08-02): fix: fix output with number groupings & add slicer in csv header
fixes: #963