mainsail-crew/mainsail #971: Filament Sensor

ckvsoft, 2022-07-16:
### Mainsail Version:

2.2.1

### Browser:

Firefox

### Device:

Laptop

### Operating System:

Linux

### What happened?

In Miscellaneous the Filament sensor display empty but QUERY_FILAMENT_SENSOR show Filament Sensor smartsensor: filament detected.

### What did you expect to happen instead?

should show detected

### How to reproduce this bug?

enable the sensor

meteyou, 2022-07-17:
please upload your klipper log

ckvsoft, 2022-07-17:
[klippy.log](https://github.com/mainsail-crew/mainsail/files/9128611/klippy.log)

meteyou, 2022-09-02:
i tested it with my test hardware and it works fine. so looks like an issue on your specific setup.

closed 2022-09-02 by meteyou

ckvsoft, 2022-09-03:
Thanks. I don't know what I could do wrong there. But good.
I disabled the sensor.
It also works in a short test. Only when the sensor has reacted and you go to resume once or twice is it simply displayed as empty, but it prints because it's okay for Klipper.