mainsail-crew/mainsail #978: hall filament width sensor

seebaer1976, 2022-07-19:
### Requested feature:

would it be possible to install the hall filament width sensor ([link to hall filament width sensor](https://www.klipper3d.org/Hall_Filament_Width_Sensor.html))?
i.e. the menu entries and the current measured filament diameter on the main screen.

Kind regards
markus

meteyou, 2022-07-19:
Would be nice to have one. It's easier for debugging and testing during the implementation.

wrutherf, 2023-03-10:
Bump - Would be nice to add "plugin" or way to interface with Klipper's Filament Sensor Width:
https://www.klipper3d.org/Hall_Filament_Width_Sensor.html

meteyou, 2023-03-12:
until now, i don't have a sensor to test this function... is there some "all in one set" available?

wrutherf, 2023-03-12:
Nothing "off the shelf" there are several width sensor models in various
sites (thingiverse, printables).

Typically housing with 2 hall effect sensors, that measure magnetic field
from magnet on sliding lever that is moved by the filament.
https://www.printables.com/model/57699-filament-width-sensor-hall-sensors

[image: image.png]

JNSmith1717, 2023-07-22:
I'm building a hall effect width sensor myself, anyone willing to add a way to see current filament width on the mainsail page i will send them a filament width sensor if need be

meteyou, 2023-07-22:
If you are able to test a mainsail build, this would also fit...