mainsail-crew/mainsail #990: [FR] UI for Input Shaper similar to the Heightmap Page

Christanoid, 2022-07-31:
### Requested feature:

It would be nice to eliminate the need for SSHing into the Pi, using the klipper terminal and manually editing the printer.cfg to setup Input shaper. An UI Tab similar to what Heightmap / Bed Mesh Leveling has would be a nice to have.
This UI would optimally have buttons to start the resonance testing, an area where the resulting graphs get displayed as well as a field to input the resulting values shown in the graph or maybe even automatically have it set the best determined values.

batista16, 2022-07-31:
This is a great idea!

meteyou, 2022-09-11:
there is no option to trigger these function (only the start of the input shaper test) in klipper/moonraker atm.

as a short help, you can just change the output path of the graph to the klipper_config (or klipper_logs) path. then you can display the graphs in mainsail.

paulstimpson, 2023-02-13:
I'd really like this. A UI page to trigger the input shaper test on an axis, run the python script then display the text output and resulting graph

salbang, 2023-04-12:
It would be really nice to have this feature.

agravelot, 2023-05-29:
A good alternative will be EIS: https://github.com/MagicPhoenix/EIS

Riksu9000, 2023-06-15:
> there is no option to trigger these function (only the start of the input shaper test) in klipper/moonraker atm.

Do you mean the python scripts that process the data? I imagine you wouldn't need Klipper or Moonraker for that.

For me just getting a graph by clicking a single button in Mainsail would be great and is probably a good starting point for this.

zellneralex, 2023-06-15:
I agree that would be a nice addition, if only klipper could generate the stuff inside ...
You all forget that the current Implantation invokes the need to ssh in your SBC and run the script by hand.

There are ways around with unsoportate klipper modules and scripts (one can be found in my repo) but there is nothing official that a UI like Mainsail can use.

So the first thing would be to open a post at https://klipper.discourse.group/ and either see if the main contributor of input shaper ([dmbutyugin](https://github.com/Klipper3d/klipper/commits?author=dmbutyugin) or any other klipper contributor would be interested in adding a klipper feature to generate these graphs insight klipper

smolz, 2023-07-13:
> A good alternative will be EIS: https://github.com/MagicPhoenix/EIS

The only issue I have found is that it runs on port 8080 with no option to change it.  This interferes with almost every webcam streamer

Jrich4967, 2023-09-07:
> > A good alternative will be EIS: https://github.com/MagicPhoenix/EIS
> 
> The only issue I have found is that it runs on port 8080 with no option to change it. This interferes with almost every webcam streamer

If you go into the git and search for 8080 in the main.go, you can probably change it there?