mainsail-crew/mainsail #993: Improve default perspective and readability of heightmap graph

batista16, 2022-07-31:
### Requested feature:

This is a feature request that only makes sense if everyone agrees (as most of them).. 
on the mesh view, the default view it's with the Y0 XMax closer to you.. that's a really weird prespective I think, it's hard for me to understand where's what, I have to move the thing arround until I get a 0,0 visible.. My proposal is to have it from the front of the printer, like I phisically look at it.. 
it would also be nice to default to a zoom where everything is in view, including labels.. right now I need to zoom out so I can see the axis labels.
Maybe even the panel height can be increased. 
Default view: 
[image]
Preferred view: 
[image]

Henlor, 2022-09-15:
I'm totally agree in this. 

I'm always lost when I see this graph. First thing I always have to do, it's trying to find the "good orientation" to understand it. 
And the second thing that's bores me is the Z axis Grid switching from side to side depending on the perspective orientation. I wonder if it's useful or even necessary to display it at all afterall

Bay the way, I'm on a Cartesian printer. I don't know if this as an impact on the initial orientation of the bed mesh graph

LazaroFilm, 2023-01-14:
Agreed too also, the X and Y scales could be by the edge of the bed instead of all the way at the bottom. The Z scale could be a toggle option.

meteyou, 2024-09-10:
@batista16 i created a PR to implement this feature. pls check it if it would fit for you.

@Henlor
> And the second thing that's bores me is the Z axis Grid switching from side to side depending on the perspective orientation. I wonder if it's useful or even necessary to display it at all, afterall.

currently the grid has to jump left and right, otherwise it covers the bed_mesh. but this is generally the default setting of the chart lib we use (echarts). i haven't seen an option to disable this yet. but we could probably make the z-grid less “strong/bright” to make it look “lighter”. maybe this would make it easier to orientate?

@LazaroFilm
> Agreed too also, the X and Y scales could be by the edge of the bed instead of all the way at the bottom. The Z scale could be a toggle option.

i dont know any option to move the labels at the height from the mesh in this lib...

PR #2006 (2024-09-24): feat(Heightmap): add option to set the default orientation
## Description

This PR adds an option in the Settings to let the user set the default orientation of the bed mesh visualization.

## Related Tickets & Documents

fixes #993 

## Mobile & Desktop Screenshots/Recordings

Option "Right Front" (default option and the same orientation as before):
[image]

Option "Left Front":
[image]

Option "Front":
[image]

Option "Top":
[image]

## [optional] Are there any post-deployment tasks we need to perform?

none
closed 2024-09-24 by meteyou