mainsail-crew/mainsail #995: "Extruder not hot enough" in mainsail.cfg's _TOOLHEAD_PARK_PAUSE_CANCEL macro also triggers when filament runout sensor reports no filament

XPav, 2022-08-02:
### Mainsail Version:

2.2.1

### Browser:

Edge

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

When homing a CR-10 with a filament runout sensor, but with no filament , the console reports: "action:paused", followed by "Extruder not hot enough", and stops.

The extruder wasn't heated at all, so this was confusing.   

I tracked it down to the default _TOOLHEAD_PARK_PAUSE_CANCEL macro, called because the G28 home issues a pause command, which issues the "Extruder not hot enough" message when "printer.extruder.can_extrude" is false.  

This is likely being triggered by the filament runout sensor in this case, not the extruder temperature.

As the main thing being worked on was the thermistor in this case, it was very confusing to figure out why this error was coming up.

### What did you expect to happen instead?

I expected the G28 to complete properly.

On a G28, I not sure the standard PAUSE command makes sense to do.  The head should be parked,  but I'm not sure this should have anything to do with the extruder or filament runout sensor at all.  

There very well could be some good reasons to do it this way, but in that case, I'd expect for the error message to be "no filament", rather than "extruder not hot enough".

### How to reproduce this bug?

Perform a G28 on a system with a filament runout sensor, but with no filament.

XPav, 2022-08-02:
[klippy (1).log](https://github.com/mainsail-crew/mainsail/files/9244132/klippy.1.log)

th33xitus, 2022-08-02:
This does not seem like a bug with Mainsail. Please only use the issue tracker for actual Mainsail related bugs.

Your filament runout sensor is set to pause on runout. That will call the PAUSE macro. Your PAUSE macro will call the _TOOLHEAD_PARK_PAUSE_CANCEL macro. And in there `printer.extruder.can_extrude` is checked.

> I tracked it down to the default _TOOLHEAD_PARK_PAUSE_CANCEL macro, **called because the G28 home issues a pause** command, which issues the "Extruder not hot enough" message when "printer.extruder.can_extrude" is false.

So this conclusion doesn't sound right. 
A G28 on its own will never issue a PAUSE command, unless you overwrite G28, which you didn't.

closed 2022-08-02 by github-actions[bot]

XPav, 2022-08-02:
Taking this to discord -- regardless of why the system is going into pause mode, mainsail.cfg's default macro reports incorrect information to the user.