mainsail-crew/mainsail #1016: Support for old iOS devices

thetouchablegod, 2022-08-09:
### Requested feature:

Proper page rendering on old iOS 9.3.5 devices.

### Solves the following problem:

Trying to manage 3 separate instances from my old first gen iPad running iOS 9.3.5 with Javascript enabled. Each instance gives a black page with nothing else on it. Unsure of the scope of what would need to be changed.

lixxbox, 2022-08-10:
Hello and thank you for the feature request. 
Software dependencies do not allow us to support iOS9/Safari 9. Unfortunately, using a different browser doesn't help either, since on iOS, all browsers have to use the same (very old) WebKit engine.
Mainsail requires at least iOS12. So you would have to use a more recent device if you are not able to update your current one.

Best regards

closed 2022-08-10 by lixxbox

Al1sa, 2026-08-29:
I've vibecoded a lightweight web UI using Cursor AI that runs on Windows PC. It talks to Moonraker on the Raspberry Pi and shows temperatures, print status, controls, and the webcam. The page uses plain HTML and old-school JavaScript that my iOS 10.3.3 can actually handle.
I have no idea how to upload such things to Github without exposing anything or with actual working instructions on how to install it. I just wanted to say that this is very possible to use an old iPad with new Mainsail, vibecoding it takes like an hour for someone who has no idea how to code.

[image]