mainsail-crew/mainsail #1024: Limit temperature changes mid-print to avoid going cold because of a typo

drelich, 2022-08-13:
### Requested feature:

At the moment the Mainsail UI disables the dropdown options for picking pre-set temperatures while a print is in progress. I would also like to see that the free text field, which is still editable during print, sets a minimum temperature allowed minus 20°C for example (usually lowering temp by 10°C is sufficient?). This could be based on the temperature found in gcode.

Had an 18 hour print fail because instead of 230°C I managed to only type 30°C on my iPad mini and didn't notice until the nozzle start making screeching noises. A print forced abort followed shortly after. That print was 53% done...

### Solves the following problem:

Prevents too low temperatures to be set mid-print, avoiding aborting a print because of a typo, which is very frequent on mobile devices.

th33xitus, 2022-08-13:
Which Mainsail version do you currently use?

drelich, 2022-08-13:
v2.2.1 (always keeping it at its latest :) )

th33xitus, 2022-08-13:
In prior versions of Mainsail, temperatures were automatically set once the input field was left. This often led to unintentional changes to the printing temperature. Starting with version 2.2.0 users are required to explicitly confirm setting a new temperature by pressing enter. Making the user confirm the changes by pressing enter is our approach to prevent such issues. On the other hand we expect the user to not change such potentially crucial parameters on the fly without paying attention before confirmation.

> This could be based on the temperature found in gcode.

We are not able to look into the gcode to see which temperatures are used during a print.

drelich, 2022-08-13:
> We are not able to look into the gcode to see which temperatures are used during a print.

How do you get the temperature that is displayed in the text field? I thought this is pre-filled from gcode during streaming?

On a small device, editing a free text is a little 'clumsy' with the limited screen estate, which is half covered with on-screen keyboard. I did confirm with Enter but didn't see the first digit was missing.

In any case, that temperature field seems to be a NUMBER type HTML INPUT. This field has the option to set MIN and MAX values. Can it be updated to come with a MIN equal to something sensible, say 180°C (I presume nobody prints lower than 180, not even with PLA?), and have validation in place to warn the user if they tried to go lower by mistake?

th33xitus, 2022-08-13:
> How do you get the temperature that is displayed in the text field? I thought this is pre-filled from gcode during streaming?

This is a fundamental misunderstanding on how Mainsail works. No gcode is streamed. Temperatures are made available through Moonraker.

>I did confirm with Enter but didn't see the first digit was missing.

[image]
I checked the layout of an iPad Mini with the developer tools in my browser to see how bad it actually is. As you can see the 230 are fully visible. This is actually something we try to pay attention to and don't let input to be cut of. So in every input you should always and in every resolution be able to see the full "sane" input.

> Can it be updated to come with a MIN equal to something sensible, say 180°C (I presume nobody prints lower than 180, not even with PLA?), and have validation in place to warn the user if they tried to go lower by mistake?

I can forward that request to the rest of the team so we can discuss that as it requires to put some business logic in place where (imho) no business logic (in form of code) should be required. I think this is more of a individual one-time issue due to some inattention and from now on you'll probably check twice before confirmation.

drelich, 2022-08-13:
Thanks. Dev tools in Chrome don't really give you the whole picture though. Things like on-screen keyboard, for example. The keyboard is huuuge by default. And compared to it, the input field's text is really tiny. If you're using the iPad mini as a touchscreen for your printer and control it from about two feet away, you will have a problem reading it. 

[image]

You say there's no business logic and that I will now check twice. Fair enough. But a minor change to the INPUT field could save somebody a multi-day print and a ton of filament the first time they make the same mistake...

JonathanHeinz, 2022-08-15:
What speaks against a simple confirmation dialog, which can be activated by option?

meteyou, 2022-08-23:
@JonathanHeinz it is already a "double check workflow". you have to edit the value + press enter. without "confirm" the value with the enter key, mainsail will redo the old value.

drelich, 2022-09-07:
I can see the 'Not Addressing' tag on this one so closing it.

closed 2022-09-07 by drelich