mainsail-crew/mainsail #1032: Exclude object feature doesn't work with Cura5.1.0

SHKinsem, 2022-08-17:
### Mainsail Version:

v2.2.1

### Browser:

Edge

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

According to the [doc](https://docs.mainsail.xyz/features/exclude_objects)
There is a line says

> Cura & Ideamaker label objects by default in the G-Code, and they are automatically compatible without any further settings.

I tried both Prusa2.4.2 and Cura5.1.
The function works great with the files sliced with prusa which proofs that there are no mistake with my setup. 
But once I try to use the file sliced by cura, no 'exclude obejct botton' shows up.
So I think it is a bug that mainsail doesn't recognise the format labeld by Cura5.1

### What did you expect to happen instead?

The 'exclude object botton' would show up while using the files sliced with cura

### How to reproduce this bug?

Use files sliced with Cura5.1

### Additional information:

I am not sure if this function would work with earlier version of cura.

th33xitus, 2022-08-17:
The post processing of the file is done either by Moonraker or by an external post processing script. If the extra data is not added to the gcode and no exclude object button appear, it is not a Mainsail bug. Please doublecheck if the new Cura is really not missing a setting. And in case you think it is a bug, open an issue here: https://github.com/kageurufu/preprocess_cancellation or in the Moonraker repository.

closed 2022-08-17 by th33xitus

SHKinsem, 2022-08-18:
OK, I will chech that. Thanks.

> The post processing of the file is done either by Moonraker or by an external post processing script. If the extra data is not added to the gcode and no exclude object button appear, it is not a [...]