mainsail-crew/mainsail #1033: Add more camera settings

thbl, 2022-08-17:
### Requested feature:

Add v4l2-ctl settings to mainsail like this octoprint plugin https://github.com/The-EG/OctoPrint-CameraSettings

### Solves the following problem:

More control of camera settings like brightness and exposure, could give better image quality.

thbl, 2022-08-17:
would also make it possible to control PTZ web cameras

meteyou, 2022-08-17:
there have to be more/better support in crowsnest or moonraker for it, but it would be possible. i added it to vNext, so we will check it in the future to implement it. thx for your request.

simonvez, 2022-10-24:
hi.. has this been implemented ? :)

meteyou, 2022-10-24:
@simonvez atm you can only use the crowsnest config to add v4l2-ctl settings

meteyou, 2024-07-14:
Because Mainsail is only a web interface and has no part on the host itself to control a camera, I close this FR until there is a software part on the raspberry pi/host that can do this and we can communicate with it.

If someone finds a way/software, please post it here and I will reactivate this FR.

closed 2024-07-14 by meteyou