mainsail-crew/mainsail #1036: Confirmation for updates

Norbs12, 2022-08-20:
### Requested feature:

Add confirmation prompt before updating anything from the update manager.

th33xitus, 2022-08-20:
> Prevents accidental updates of Klipper that require flashing your MCUs.

How? Please elaborate on how a confirmation prompt will prevent those "accidential updates".

Norbs12, 2022-08-20:
Not trying to be rude but I don't understand how it's a difficult concept. There an option for confirming an emergency stop. It gives you an extra moment to think about what you're about to do or to avoid an accidental click from ruining your print. This is no different.

I generally update everything except Klipper just because the MCU updates are a hassle. Every other update is usually safe. I accidentally clicked the wrong thing and now I spent a bunch of time opening my printer to update the MCUs, Avoided the update successfully for over a year and keeping everything else up to date. Knew instantly I screwed up after I pressed update all, but it was too late, No way to cancel, no delay, no confirmation. It was done.

Does that make sense?

meteyou, 2022-08-20:
We only add confirmation for buttons which will kill the current print. Updates are not available during a print.

closed 2022-09-10 by github-actions[bot]