mainsail-crew/mainsail #1038: Temperature input font size too small

mikhailbot, 2022-08-20:
### Mainsail Version:

v2.2.1

### Browser:

Safari

### Device:

Mobile device (Phone, Tablet or similar)

### Operating System:

iOS

### What happened?

In Safari on iOS font sizes smaller than 16px causing the browser to zoom in. Often times this causes the interface to gain a horizontal scroll causing alignment and usability issues.

### What did you expect to happen instead?

No zooming in, keeping viewport consistent. 

### How to reproduce this bug?

Visit the web ui on iOS with Safari and edit one of the temperature inputs.

meteyou, 2022-08-23:
pls post a screenshot

mikhailbot, 2022-08-23:
[image]

See attached. It's not *major* but it is common practise to not have input fields smaller than 16px to prevent the zoom-in by Safari on iOS (which then adds horizontal scrolling requiring you to zoom out but not too far or you go into tab switching mode or reloading the page).

Of note I've another install on 2.1.2 without the smaller font size.

meteyou, 2022-08-23:
do you use a theme or something else? it cannot be only in one installation an issue...

mikhailbot, 2022-08-23:
Nope, plain iOS and plain MainsailOS install and for reference it's this CSS too.

https://github.com/mainsail-crew/mainsail/blob/develop/src/components/inputs/TemperatureInput.vue#L3

PR #1107 (2022-10-05): fix(UI): tweak font sizes
This PR tweaks a few font sizes, primarily the ones of the temperature input field and the slider input fields so they are 1rem / 16px tall. This should fix #1038

The font of the state message was increased to 14px, so its identically to the fontsize of the current temperature.
Also the font size of the presets accessible via the dropdown to the right hand side of the temperature input fields were increased to 14px.

## Before:
[image]

## After:
[image]
closed 2022-10-05 by th33xitus