mainsail-crew/mainsail #1044: Turkish Traslation of Mainsail

Liechtenstein1, 2022-08-25:
### Requested feature:

I have successfully translated Mainsail into Turkish, can you please add the traslation as soon as possible? (.json and babel file is in the attachment)
[tr.zip](https://github.com/mainsail-crew/mainsail/files/9422939/tr.zip)

th33xitus, 2022-08-27:
Thanks for providing the translation. We don't use babel files. The JSON-file is enough.
This time i created a PR for the initial implementation of the Turkish translation. Are you able to provide further translations in the future yourself by using Github Pull Requests? If you're on our discord server, feel free to contact one of our community managers in case you are interested in receiving a translator-role.

meteyou, 2022-08-27:
thx for your translation. i just found 2 missing keys. would it be possible to post the translation here?

[image]

th33xitus, 2022-08-28:
I updated the translation with what i think might be correct:
[`8ad4cb5` (#1049)](https://github.com/mainsail-crew/mainsail/pull/1049/commits/8ad4cb5965f46ffe3186f4abd44d15db58a7330e)

The first one was already translated in 'Files.DeleteSelectedQuestion", so i took it from there.
A confirmation for the second one would be nice though as i just used a translator for the word "edit".
Looking at other keys where "edit" was translated, it might be correct though.

Liechtenstein1, 2022-08-28:
"Do you really want to delete {count} selected files?" Shoud be 
"Seçilen {count} dosyayı gerçekten silmek istiyor musunuz?"

and

"Edit crowsnest.conf" should be "crowsnest.conf'u düzenle"

And yes I am interested in a translator role but i don't have enough experience using GitHub to creating PR request's. I will join to discord server in next Thursday as soon as I get back from my annual holiday, thank you for your interest

th33xitus, 2022-08-28:
I updated the PR as per your suggestions. Thanks for that!

Okay, just contact us once you joined our Discord :)

closed 2022-08-28 by th33xitus

efyland, 2025-01-30:
> ### Requested feature:
> I have successfully translated Mainsail into Turkish, can you please add the traslation as soon as possible? (.json and babel file is in the attachment) [...]

Bazı kahramanlar pelerin takmaz. teşekkür ederiz <3
Some heroes don't wear capes. thank you &lt;3

PR #1049 (2022-08-28): feat: add Turkish localization
fixes #1044 

Thanks @Liechtenstein1 for providing the translation!