mainsail-crew/mainsail #1054: G Code Viewer Progress Color Mode

raimonds1503, 2022-09-01:
### Requested feature:

Add new Color Mode to G-Code Viewer that has 2 colors. One for the layers that has already been printed and one for the ones that are not yet to be printed.

### Solves the following problem:

This makes it easier to see how far of a model the print has been done, to spot features that are upcoming. Tracking does not help with this, as it hides the model thats not yet printed. And its sometimes hard to spot the nozzle when the entire model is visible.

meteyou, 2022-09-07:
for have a good working solution, we would need layer numbers from klipper. we also started a PR for this features in klipper: https://github.com/Klipper3d/klipper/pull/5726

Sindarius, 2023-01-06:
For now I recommend you can use the transparency option. Parts that are printed are solid and the unprinted will be transparent giving a similar effect

maruhe, 2023-03-16:
could you pls make it a bit more transparent? 🤪🧐
or can I change the value somewhere?

https://user-images.githubusercontent.com/49123648/225700769-36e5c365-e460-43f1-961c-b30af2205f0b.mp4