mainsail-crew/mainsail #1063: G-Code viewer - slider to browse layers

konwin, 2022-09-07:
### Requested feature:

Hello,
first thank you for this great tool. In comparison to ocroprint mainsail runs fast as hell!
I actually miss only one thing. Could you implement an option to switch the slider in the G-Code viewer to browse whole layers?
Atm it is showing every single character of the g-code file, right? What sense does this make?
[image]
[image]
I searched everywhere for a documentation, but couldn't find one regarding the g-code viewer.
Regards
Konrad

meteyou, 2022-09-07:
for have a good working solution, we would need layer numbers from klipper. we also started a PR for this features in klipper: https://github.com/Klipper3d/klipper/pull/5726

Exeu, 2023-05-29:
This would be nice!

meteyou, 2023-05-29:
ok. the klipper dependencies are exists now. because we only use this gcode viewer in Mainsail, this feature have to be integratet in the gcode viewer first, that we can use it it: https://github.com/Sindarius/npm_gcodeviewer

PhilBaz, 2023-07-17:
Oh awesome...this will be a huge improvement! The gcode viewer looks great, but without layers its just not very useful.