mainsail-crew/mainsail #1066: Would love to always hide "SAVE CONFIG" button.

HyperToxic33, 2022-09-10:
### Requested feature:

My printer is on all the time 24/7 - Bed Mesh is part of my PRINT_START.   SAVE CONFIG is always shown and I never know what it would save, because many thing trigger the display of SAVE CONFIG.  

Bed Mesh among other things causes SAVE CONFIG button to be shown and enabled.   There is currently no way to know what values will be changed when SAVE CONFIG is pressed.   I do not ever want to use this feature because of the ambiguity of what may be changed.

This best solution is to provide a toggle button in the config that hides the SAVE CONFIG button all the time.  That way I don't need to worry about accidently pressing the button and modifying my config file.

meteyou, 2022-09-11:
> I do not ever want to use this feature because of the ambiguity of what may be changed.

this is not correct. this PR in klipper (<https://github.com/Klipper3d/klipper/pull/5270>) provide all new/changed things. so it is on my todolist to add this feature.

but also with this implemention. i would add some kind of "filter". so you can only hide the button for bed_mesh changes. i have the same issue on my switchwire. i create a new mesh at the beginning of every print.

LazaroFilm, 2023-01-14:
One thing I'm not sure is understood. When you press SAVE_CPNFIG, it saves all the changes that were made to the bottom section of your printer.cfg file. It doesn't only save the last change made, but any and all scheduled change. But it only changes values in that bottom section, not throughout your Config. 
Also, it creates a backup of your Config file in its previous state.

PR #1255 (2023-02-05): feat: add button to hide SAVE_CONFIG button for pending bed_mesh
this PR fixes #1066 

screenshot deactivated:
[image]

screenshot activated:
[image]
closed 2023-02-05 by meteyou