mainsail-crew/mainsail #1073: Unexcluded from exclude object - currently no way to correct a mistake.

PhilBaz, 2022-09-13:
### Requested feature:

It would be great if objects that have been excluded could be re-included. The only reason is to correct a mistake that has just been made. 

### Solves the following problem:

If you accidently chose the wrong object, that's it, game over. There is currently no way to correct the mistake.

meteyou, 2022-09-13:
klipper has no support to "re-include" an object. so this is not possible to include in the gui, when the firmware have no support.

closed 2022-09-13 by meteyou

PhilBaz, 2022-09-13:
@meteyou Ahhh, damn. Yeah, I figured this one out the hard way.

meteyou, 2022-09-13:
@PhilBaz but i personally don't find this feature useful. you would have to reactivate the object immediately. as soon as 1 layer is missing, it could even ruin the print.
i would also have no possibility in mainsail to figure out whether the object has already been skipped or not. thus it would also not be possible to display it only in mainsail until the next layer has started.

PhilBaz, 2022-09-13:
Its more of an oh shit button then a real feature. I excluded the wrong object. Could have saved, its layer wasn't coming up for a solid minute. But it was game over on click.

If it wouldn't play nicely with UI fair enough...there is just no possibility to correct a mistake now.  Say you have a plate with 8 items. can be tricky.

meteyou, 2022-09-13:
you cannot identify it on the map or in the gcode viewer?

PhilBaz, 2022-09-13:
@meteyou ...On the map. I mean I can identify it, I can also mess it up too, apparently. Say I'm doing a tray full of Voron parts on my 350^3,... maybe I pick the right square, maybe not.

poon-holder, 2022-09-15:
I've not engaged with the exclude feature yet, however would a bigger flashier "Are you sure?" prompt not be easier? 
After that it's kind of down to us isn't it ? :)

meteyou, 2022-09-15:
> I've not engaged with the exclude feature yet, however would a bigger flashier "Are you sure?" prompt not be easier? After that it's kind of down to us isn't it ? :)

There is already a prompt

poon-holder, 2022-09-15:
Sorry, we all know sarcasm doesn't bode well in text, I meant one with a "trigger warning" so the flakes know what they're getting into if they cant fathom the complexity of their desires. 
A small graphic of a cup for their tears might gel them up. :)
The exclude feature is a god send I thought was only a pipe dream, this request is audaciously obscene!!!

PhilBaz, 2022-09-15:
@poon-holder 

Its an off switch...I'm requesting it can be turned back on. That's all.