mainsail-crew/mainsail #1074: Download multiple config files from machine page.

PhilBaz, 2022-09-13:
### Requested feature:

The ability to download multiple config files from machine page. 

### Solves the following problem:

Needing to click each file when backing up all config files.

meteyou, 2022-09-13:
thx for this FR. sounds reasonable to me. i added it to the backlog.

PhilBaz, 2022-09-13:
Awesome...thanks.

Zodiak1993, 2022-09-29:
It would also be very handy, if you hit backup under settings in mainsail, that there is also a backup of the config path. Maybe selectable ?

I don't know if it is possible to create a .zip or .rar for this files.

meteyou, 2022-09-29:
@Zodiak1993 i think for this i should talk to arksine that moonraker can offer a zip.

PR #1194 (2023-01-07): feat: add multi download to ConfigFilesPanel.vue
fixes #1074
closed 2023-01-07 by meteyou