mainsail-crew/mainsail #1076: Add contrasting text color for RESPOND TYPE=command (  PREFIX="// " )

Henlor, 2022-09-13:
### Requested feature:

Add a text color to command type console messages 

RESPOND TYPE=command MSG="Hello World"
RESPOND PREFIX="// " MSG="Hello World"
RESPOND MSG="// Hello World"

Henlor, 2022-09-16:
This issue was resolved with this macro. 

https://github.com/Henlor/klipper_config_CR-6_SE/blob/65320b545e6f849a8874849be2db70b8647ffe18/macros/respond.cfg

closed 2022-09-16 by Henlor

livinhack, 2022-09-22:
Resolved? I get this error: Existing command 'RESPOND' not found in gcode_macro rename

Mainsail 2.3.1

th33xitus, 2022-09-22:
This has nothing to do with Mainsail

livinhack, 2022-09-22:
His macro should work with mainsail so it's meant as additional info that I'm using it.

But what am I missing?

Henlor, 2022-09-22:
> I think so too.
> 
> His macro should work with mainsail so it's meant as additional info that I'm using it.
> 
> But what am I missing?

You need to register the [respond] command first if you want to extend it with this macro
https://www.klipper3d.org/Config_Reference.html?h=respond#respond