mainsail-crew/mainsail #1085: Expose mainsail host command 'shutdown'  to console.

PhilBaz, 2022-09-15:
### Requested feature:

Perhaps this breaks the only klipper features rule.....but

It would be nice to have access to the shutdown command through a macro/console. Its my understanding shutdown is a moonraker rather than klipper process. 

If accessible through macros it would allow for some very useful options. i.e. shutdown through mini screen menu, print_end etc.

### Solves the following problem:

No way to fully shut down except for using the GUI button. M112 isn't the right command as it doesn't shutdown the pi

th33xitus, 2022-09-15:
You can use the gcode_shell_command module for klipper (installable via KIAUH or downloadable from the KIAUH repository) and just write a corresponding shell command for that, then call that shell command with a macro.

There is no other way i can think of. Everything Mainsail has access to (in regards to the host system), has to be provided by Moonraker, and must go through Moonraker.

closed 2022-09-15 by th33xitus

PhilBaz, 2022-09-15:
@th33xitus 

Ahh, okay...Ill look into this! Thank you very much!

meteyou, 2022-09-16:
you can also use just a macro and the moonraker action to execute the shutdown:

```
[gcode_macro SHUTDOWN]
gcode:
  {action_call_remote_method("shutdown_machine")}

[gcode_macro REBOOT]
gcode:
  {action_call_remote_method("reboot_machine")}
```

with this, you dont have to install anything.

PhilBaz, 2022-09-16:
@meteyou 

Ahh, fantastic!

Thank you very much!