mainsail-crew/mainsail #1093: Failed to update system packages

kimprints, 2022-09-19:
### Mainsail Version:

2.3.1

### Browser:

Edge

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

update fails when trying to update system packages through the web UI
[image]

### What did you expect to happen instead?

for the update to finish

### How to reproduce this bug?

click update on the mainsail update page/widget

th33xitus, 2022-09-19:
The error comes from your linux system and is very likely not caused by Mainsail.
You can try to manually update the system through SSH with the command line.
Assuming you have MainsailOS or another Debian based image installed, the command for that is
`sudo apt-get update && sudo apt-get upgrade -y`

kimprints, 2022-09-19:
i'll try that. I also read from an old post that this could've been caused when using sudo during the initial installation. In any case,  i'll try to update through the cmd line. thank you!

th33xitus, 2022-09-19:
Okay. Im going to close this issue for now. If you still need help or have questions, feel free to join our Discord server and we can help you there.

closed 2022-09-19 by th33xitus