mainsail-crew/mainsail #1097: Mainsail can't handle 1080p webcam stream

thbl, 2022-09-23:
### Mainsail Version:

v2.3.1

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

When using a 1080p webcam stream, the video is lagging alot on mainsail's dashboard.
-If i open same stream directly or in Fluidd dashboard it runs smooth

So seems to be some kind of issue with the video embedding code ?
If i set the stream to 720p@30fps it runs fine, but when i set 1080p@30fps it lags alot.

### What did you expect to happen instead?

Showing the stream as smooth as in Fluidd dashboard

### How to reproduce this bug?

set a webcam stream to 1920x1080 @ 30fps and add it to mainsails dashboard

meteyou, 2022-09-23:
Which method do you use?

thbl, 2022-09-23:
> Which method do you use?

MJPEG

meteyou, 2022-09-23:
There are 3 mjpeg modes in mainsail

closed 2022-10-15 by github-actions[bot]