mainsail-crew/mainsail #1102: Bad Mobile UI - Emergency stop button looks like an alert

icecaster, 2022-09-30:
### Mainsail Version:

v2.3.1

### Browser:

Chrome

### Device:

Mobile device (Phone, Tablet or similar)

### Operating System:

Android

### What happened?

When loading the mainsail UI on a small screen (mobile), the "Emergency Stop" button appears like an alert message.

[image]

### What did you expect to happen instead?

To see a notification with printer errors / warnings

Instead the print is cancelled

### How to reproduce this bug?

Open the mainsail UI on a small screen

meteyou, 2022-09-30:
I close this issue, because a misinterpretation of an icon is not a bug. The emergency stop button works and the notification button also work. so no function have a bug.

if you have an idea that is self-explanatory for ALL users, feel free to open a FR or make a PR.

closed 2022-09-30 by meteyou

fritzw, 2023-08-17:
@meteyou The same happened to me a few times over the years, and my wife also did the same just now. So it seems to be a real problem, whether or not you classify it as a bug.

Something similar to this could be a more fitting design for the button: https://commons.wikimedia.org/wiki/File:Stop_hand-octogon-red.svg

I would guess that both the octagonal shape and the lifted hand would signify "STOP" to many people. No way to confuse it with a notification/alert button.

meteyou, 2023-08-17:
@fritzw we switched it multiple times until now and everytime a new user come and say the same thing. So I don't want to change it again.

Feel free to enable the prompt if you think you will misclick it again...