mainsail-crew/mainsail #1123: Preferences option to allow "tab-key" to admit value

Shteeeeeve, 2022-10-17:
### Requested feature:

Preferences option to allow "tab-key" to admit value for extruder / bed temp. This was the default behaviour in previous releases and was removed to avoid unintentionally changing the temperature. A preferences option to allow both methods based on user preference.

### Solves the following problem:

Align to previous functionality that users may be used to / prefer.

FatalBulletHit, 2022-10-23:
This might also be related to the X, Y and Z control of the Toolhead section where pressing tab reverts to the current position.

It is probably preferable if pressing tab was to switch to the next field (without reverting to the current position/temperature) and doing a combined move/temperature adjustment when enter is pressed, e.g.:

`input 175 for X -> tab -> input 175 for Y -> tab -> input 50 for Z -> enter -> G1 X175 Y175 Z50`

`input 100 for X -> tab -> tab -> input 10 for Z -> enter -> G1 X100 Z10`

`input 245 for extruder target -> tab -> input 110 for bed target -> enter -> SET_HEATER_TEMPERATURE HEATER=extruder TARGET=255 -> SET_HEATER_TEMPERATURE HEATER=heater_bed TARGET=110`

Additionally, delayed reverting to the current position (or temperature) could be used for safety reasons, e.g.

`input 175 for X -> tab -> input 175 for Y -> no input for 10 seconds -> revert to current position`

`input 245 for extruder target -> tab -> no input for 10 seconds -> revert to current target`

matthewdolman-afd, 2022-10-26:
I would also appreciate an option to return to previous behaviour.  Having to remember to press enter on each temperature change is really irritating.