mainsail-crew/mainsail #1124: Add Move option to GCode Files

officiallymarky, 2022-10-19:
### Requested feature:

Right now you can create folders under GCode Files, but as far as I can see, there is no easy way to move existing GCode files into that folder.  It would be nice to be able to move files into folders from the UI without downloading and re-uploading.

Something like right click Move option would be awesome.

### Solves the following problem:

Organizing GCode files

meteyou, 2022-10-19:
Drag & drop is implemented to move gcode files.

closed 2022-10-19 by meteyou

officiallymarky, 2022-10-19:
> Drag & drop is implemented to move gcode files.

I was trying to do that, and I was seeing the + sign as I did it.  Some reason it wasn't working.  I tried again and got it working but I tried 4-5 times before and it just didn't do anything.  I'll keep an eye out and see if I have problems later.

officiallymarky, 2022-10-24:
Here is an example gif, it works sometimes but not always.  In fact, most of the time for me it doesn't seem to work.

[image]

NameOfTheDragon, 2024-12-31:
The drag-and-drop functionality is rather limiting and (according to reports) somewhat broken. How do I move multiple files to a sibling directory, for example? Answer: you can't. There really does need to be a way to mark multiple files and move them to a directory that isn't currently visible on the display. The classic use-case is when you mis-types the directory name in the slicer when uploading files. So now you have two sibling directories with slightly different names and you want to merge the contents. If you can show me how to achieve that, then I'll shut up and go away.

meteyou, 2024-12-31:
@NameOfTheDragon https://github.com/mainsail-crew/mainsail/issues/1209