mainsail-crew/mainsail #1127: Ascending or descending arrow next to temperatures - with smooth time.

PhilBaz, 2022-10-25:
### Requested feature:

For the UI it would be very nice to have an arrow next to the temp numbers indicating whether the temp is rising or falling. 

### Solves the following problem:

With thermal compensation stuff (frame comp, virtual gantry backers etc.) The frame and gantry can take a long time to change direction from cooling to heating and vice versa. This would save having to figure out what is happening every time you want to check if the temp direction has changed yet.

### Additional information:

I think this would be a really nice and very useful feature.

meteyou, 2024-12-30:
I'm closing this FR because it's a duplicate. Unfortunately i saw it too late and discussed the feature in the newer FR.

duplicate #2098

closed 2024-12-30 by meteyou

PhilBaz, 2024-12-30:
@meteyou Sorry, I thought I had made this request before. I looked but couldn't find it. I'm not great at Github.