mainsail-crew/mainsail #1129: Incremental reminders based on number of hours printed - i.e. lubricate, change filter media etc.

PhilBaz, 2022-10-26:
### Requested feature:

Since mainsail tracks total hours printed it would be very nice and useful to have customizable incremental reminders based on number of hours printed. 

For instance, a reminder to change filter media every 50 hours of printing. Or a reminder to lubricate rails every 500 hours ect.  

### Solves the following problem:

the need to track these things which we all should do, but probably no one does... especially the filter media. 

### Additional information:

I think this would be very useful and everyone could benefit from it.

Zodiak1993, 2022-11-08:
I am already using this. Create a macro with a delayed gcode - update the delayed gcode every 5min and save the value to the sd card. Thats the way how I change my filter. After change I can reset the saved value.
But I would also prefer a feature in mainsail :)

PhilBaz, 2022-12-11:
@Zodiak1993 

Ahhh, mind posting your code for this? Yes, hope they will think its a good idea and integrate it. But until then at least I can keep track.

barecool, 2023-01-01:
would be awesome if: 

it would:
- read/ write to variables.stb - to track printer hours / filter hours / maintenance actions
- display a popup on mainsail ui with a message of required maintenance action 
    - ability to dismiss pop-up, delay, reset hour meter (variables.stb)

JEDean, 2023-02-07:
Excellent Idea!

What would you think about basing the reminders on the linear distance traveled by the specific axis or extruder rather than the hours? Hours would probably do the job. But, I wonder if travel distance would better for things like bearings are wheels.

PhilBaz, 2023-02-07:
@JEDean 

I think hours is the way to go. Especially for filter media.

meteyou, 2023-02-08:
I would use the "moonraker history totals". So Filament meters would also be possible, but axis movements are not included in these values: https://moonraker.readthedocs.io/en/latest/web_api/#get-job-totals

PhilBaz, 2023-02-08:
@meteyou 

Ahh, glad to see you like this idea. It really would be a helpful little feature.

Royal ascent!

jeremiahvpratt, 2023-02-16:
Hey @meteyou ! I'd like to try my hand at implementing this feature. My high level plan:

- New "reminders" panel on the home screen that renders a table of active reminders
- Add reminder button that pops up a dialog with "name" and "print hours" fields
- Popup that displays when the reminder's designated number of print hours have passed, with "snooze" and "dismiss" buttons; "snooze" extends the reminder for a configurable number of print hours, "dismiss" deletes the reminder
- New store leaf under /store/server/reminders that posts reminders to namespace "reminders" in the database, and retrieves them for the client. Reminders are stored with a start_total_print_time equal to the current total print time from moonraker history totals, and this is used to compute number of hours remaining in the reminder.

If this sounds un-problematic, I'll try to cobble something together over the next couple of weeks :) thanks!

PhilBaz, 2023-02-16:
@jeremiahvpratt 

Ahh awesome, thanks for picking up the feature!

One thought. Maybe a separate panel is to much?  I had always imagined the reminders would show in the existing notification area ( the bell icon).

But then, I don't understand anything about what's necessary to implement the feature. 

Thanks again! Very excited :)

jeremiahvpratt, 2023-02-16:
@PhilBaz it's probably preferable that there be _some_ way to view your currently set reminders, edit them, etc. - I'm not sure that adding those things to the notification area would be the most proper. If they don't merit their own panel, potential alternatives:
- adding them to the "miscellaneous" panel. arguments in favor would be that reminders are pretty lean in terms of info, so adding a row there to show remaining hours on a given reminder is relatively low in terms of intrusiveness
- adding them as a new table area on the "history" page. argument for that would be "it relates to the other info shown on the history page", but the logical connection isn't super strong, and I think it would make it difficult for users to discover the feature.

PhilBaz, 2023-02-16:
@jeremiahvpratt 

Ahh, I hand thought about about setting and editing them.... Good point. Ill be quiet now :)

jeremiahvpratt, 2023-02-16:
@PhilBaz I do think that having expired reminders show up in the notifications area instead of as a popup may make sense! My only qualm is that there are several different things a user might want to do when a reminder expires - set it to run again, snooze it for a few print hours, or delete the reminder altogether, and it's not clear to me what that user flow would look like from the notification area - it's a lot easier to fit a few buttons in a popup. If you have thoughts about that lmk!

PhilBaz, 2023-02-16:
@jeremiahvpratt 

I hear you. I hadn't really thought about the workflow at all, just where it might appear.  Having a panel for workflow and the reminder itself showing in the notification area might be good.

yes, functions of setting, resetting, and checking the status/run time of the reminder will all be good things to have included in terms of functions.

PhilBaz, 2023-02-16:
@jeremiahvpratt 

Might also be nice to have all reminder(s) run time show in console when a print is started or finished.  Like  'Filter: 36 Hours' etc.

meteyou, 2023-02-16:
I would have thought to do the reminder in the notifications. furthermore, I would do the "create" and "manage" of the reminder on the history page. would have thought between the stats and the history print list a separate panel for it.

jeremiahvpratt, 2023-02-16:
@meteyou sure, makes sense to me! In that case I’ll fold the reminder store code into the existing store/server/history files

meteyou, 2023-02-17:
I would use a separate store, because you also need different functions for reminders

jeremiahvpratt, 2023-03-01:
@PhilBaz yeah, it’s been a busy few weeks, hopefully will get around to addressing edits this coming weekend! If you want to try it out, my branch is in a useable state, and any reminders you make on the branch should persist if it eventually gets incorporated into a real release.

PhilBaz, 2023-03-01:
@jeremiahvpratt 

Cool, Il see if I can figure out how to get it installed :)

PR #1274 (2024-04-28): feat: Reminders panel on the History page
### This PR adds a Reminders panel, which allows users to create and manage incremental reminders based on an elapsed number of print hours since reminder creation.
This PR will fix #1129 

### How to use it:

1. Ensure history is enabled in your moonraker conf
2. In the History tab, click the "+" button in the Reminders panel
3. Add a name and print hours duration to your reminder
4. Print for that many hours, then observe a reminder appearing in notifications.
5. Repeat a reminder by right clicking it and selecting "repeat" - this resets the start ti [... cut]
closed 2024-04-28 by meteyou