mainsail-crew/mainsail #1130: On mobile the emergency stop icon looks like a notification

matthewdolman, 2022-10-26:
### Requested feature:

Change the icon of emergency stop or make it clear on mobile that it isnt a notification

### Solves the following problem:

I was getting to the end of a 16 hour print, I was wondering if it had finished but couldn't be arsed going into the other room, so bought mainsail up on my phone, I saw the exclamation mark icon and clicked it thinking it was a notification...it cancelled the print...very close to the end.  For whatever reason my brain thought that the exclamation mark meant notification instead of emergency stop.  It would be cool if the icon were a bit more expressive, maybe a 'halt' hand?

### Additional information:

Not an issue on desktop as there is accompanying text.

meteyou, 2022-10-26:
sry... this topic was more than one time and there is no icon that everyone will understand it...

i can only say: don't touch on buttons you dont know what they do, when you are printing!

closed 2022-10-26 by meteyou

th33xitus, 2022-10-26:
[image]

matthewdolman, 2022-10-26:
All I am saying is that a 'halt' hand is clearer than an exclamation mark that looks like a notification.  You think that people see a 'halt' hand and think that means 'notification'?

meteyou, 2022-10-26:
as i said. we discussed this topic multiple times (on github and in the discord). multiple requests with multiple icon requests and every time another user think its wrong... i think this is also already the 3rd icon we use... same topic as before...