mainsail-crew/mainsail #1137: Customizable mapping to dashboard "type" based on user-agent

Misterke, 2022-10-31:
### Requested feature:

Allow the creation of custom dashboard layouts (including setting the number of columns and ideally font-size or zoom factor) and the selection of what dashboard layout is used based on the user-agent header rather than only the resolution of the client.

### Solves the following problem:

Currently the dashboard layout seems to be selected purely on the resolution of the client, but I feel this is sometimes unsuitable.  In my case I have a Nexus 10 tablet, which is pretty dated, but has a pretty high resolution.  Result is that it cannot handle what my other devices with similar resolution can accept.

Better would be if we could define custom additional layouts and have those applied when the user-agent matches the configured regex.

meteyou, 2022-11-01:
I think i would only make the breakpoints configable. Setting user agent specific would be too complicated

Misterke, 2022-11-01:
> I think i would only make the breakpoints configable. Setting user agent specific would be too complicated

It can be made relatively easy if you allow a user to choose from the prepared dashboard layouts for "this device".  I.e. you first add the dashboard type on a device where performance isn't an issue, set the number of columns and so and then on the device where you want to use this layout you just navigate to the settings and can then select "the layout to use on this device" from a dropdown which by default will be "automatic" (meaning there is no binding for the user-agent of your device and hence the normal resolution-rules apply) but when changed adds a binding from your current user-agent to the selected layout.

Misterke, 2022-11-04:
Come to think of it ... Perhaps storing the selected dashboard layout in client-side storage could allow a user-friendly overruling.  You could just add a "This Device" dashboard layout and any change done to that would use layout info that is stored in the client-side storage of the device doing the configuring ...

Misterke, 2022-11-21:
I do still consider this relevant.  The title to bind stuff to the "user-agent" header might no longer fit what I mentioned in the latest comments (relying on client-side storage to identify what dashboard to use), but I do still think that the current mapping by resolution / aspect-ratio should be extended to allow a user to explicitly mark the current device to use a specific dashboard layout even if the resolution / aspect-ratio might hint at another one.

meteyou, 2022-11-21:
As I said. Editable breakpoints would be a better solution instead of useragent storage.

Misterke, 2022-11-23:
Sorry, but I do not understand what you mean with "editable breakpoints" and hence don't see how this will address my problem.  Please elaborate.

meteyou, 2022-11-23:
A "breakpoint" is a viewport width definition (in pixels) when the GUI should reorder the dashboard.

mainsail use the default breakpoints from vuetify atm: https://vuetifyjs.com/en/features/breakpoints/

Misterke, 2022-12-22:
I do not feel this "breakpoint" addresses the actual problem one has with lower performance devices.

In my case the problematic device is a Nexus 10 tablet, which has one of the highest resolutions of devices that I own (it was made by Samsung and Google at the height of the pixel-density-war), but has a CPU that is meanwhile somewhat dated and the amount of memory meanwhile also limits the performance of the browser.  So, selection based solely on resolution isn't matching what I would need to disable stuff like webcam or other more complex panels on this device.

Of course, if I could just make it fine-grained enough (i.e. exactly that tablet's reported width and height uses a particular layout and anything else does not), I could get away with this approach but the idea I was proposing is less for dynamic re-orienting of panels in the dashboard rather than really being able to disable stuff based on the capabilities and limitations of the device the GUI is rendered on, possibly not limited to just the dashboard, but also making other sections less "heavy".

meteyou, 2022-12-23:
A nexus 10 tablet has a viewport size of 1280x800 (viewport size != resolution). So you would only have to enlarge the tablet area and have it cleanly separated from your desktop.

Furthermore, you are the only user who has contacted me about this topic, so I see a too complicated way of implementing it as a not reasonable solution.

Misterke, 2022-12-23:
I do not know what you consider "viewport size", but with a resolution of 2560×1600 it seems strange that would be 1280x800.  Anyway, I'm going to close this item as it seems clear you aren't willing to add any customization based on the device itself rather than the resolution or display size.  I'll probably just end up making my own low-complexity web-GUI similar to KlipperScreen, but in the browser or might continue just using VNC on the tablet to view KlipperScreen that runs on a vncserver on the pi (as running an X server on the tablet itself also doesn't work).  Anyway ... closing ...

closed 2022-12-23 by Misterke