mainsail-crew/mainsail #1141: Watchdog warning on Moonraker due to high load by Mainsail

Misterke, 2022-11-04:
### Mainsail Version:

v2.3.1

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Other (Please tell us below)

### What happened?

When opening the Mainsail interface it "initializes" the connection, but while doing that it seems Mainsail is pulling in information on all files (recursively) in certain folders of the system.

As I have my klipper config in a git repo, I see stuff like this when enabling verbose Moonraker logging:
```
2022-11-04 15:46:43,980 [websockets.py:dispatch()] - Websocket Received::{"jsonrpc":"2.0","method":"server.files.get_directory","params":{"path":"config/.git/objects/b1"},"id":8971}
2022-11-04 15:46:43,985 [websockets.py:dispatch()] - Websocket Response::{"jsonrpc": "2.0", "result": {"dirs": [], "files": [{"modified": 1649232612.6807237, "size": 119, "permissions": "rw", "filename": "4f6635691e151c79a443ed8b3ab24e41a5baa5"}], "disk_usage": {"total": 62732877824, "used": 44894171136, "free": 15254269952}, "root_info": {"name": "config", "permissions": "rw"}}, "id": 8971}
2022-11-04 15:46:43,985 [websockets.py:dispatch()] - Websocket Received::{"jsonrpc":"2.0","method":"server.files.get_directory","params":{"path":"config/.git/objects/d4"},"id":9408}
2022-11-04 15:46:43,991 [websockets.py:dispatch()] - Websocket Response::{"jsonrpc": "2.0", "result": {"dirs": [], "files": [{"modified": 1649232600.8407106, "size": 1938, "permissions": "rw", "filename": "514594e62ace78a983c3a0e24d7ca2533d7351"}], "disk_usage": {"total": 62732877824, "used": 44894179328, "free": 15254261760}, "root_info": {"name": "config", "permissions": "rw"}}, "id": 9408}
```

Needless to say that this recursive fetching can cause lots of load and slows down the entire system.  As I also have a "tmp" folder in my Timelapses folder, where lots and lots of JPEG snapshots can accumulate, those further aggravate the problem.

I don't see why Mainsail would have to ever recursively prefetch stuff.  Hence, I would strongly suggest removing this recursive fetching!

### What did you expect to happen instead?

No recursive fetching of file information and speedy startup.

### How to reproduce this bug?

I would expect that just creating a subfolder in your config folder with loads and loads of small of empty files would allow reproduction.

meteyou, 2022-11-04:
This is needed for many functions...

closed 2022-11-04 by meteyou

Misterke, 2022-11-04:
> This is needed for many functions... And these are only metadata and not the files itself.

Why would any function of mainsail need the contents of my .git folder and especially at startup?  With my tmp folder filled with JPGs any opening Mainsail on a browser took up to 10 minutes.  Moonraker would complain in its log about its event loop being blocked and queuing up actions up to a point where I think even the websocket would get closed because things were getting delayed too much:
```
2022-11-04 13:54:40,436 [file_manager.py:get_file_list()] - Updating File List <gcodes>...
2022-11-04 13:56:59,187 [app.py:log_request()] - 101 GET /websocket (127.0.0.1) [_TRUSTED_USER_] 6.63ms
2022-11-04 13:56:59,188 [websockets.py:open()] - Websocket Opened: ID: 1707607184, Proxied: True, User Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/107.0.0.0 Safari/537.36, Host Name: octopi.lan.quintux.com
2022-11-04 13:56:59,216 [websockets.py:_handle_identify()] - Websocket 1707607184 Client Identified - Name: mainsail, Version: 2.3.1, Type: web
2022-11-04 13:57:01,574 [proc_stats.py:_watchdog_callback()] - EVENT LOOP BLOCKED: 4.06 seconds, total blocked count: 3
2022-11-04 13:57:02,550 [proc_stats.py:log_last_stats()] -
System Time: 1667566615.931586, Usage: 1.94%, Memory: 29576 kB
System Time: 1667566616.934498, Usage: 1.17%, Memory: 29576 kB
System Time: 1667566617.954353, Usage: 2.49%, Memory: 29576 kB
System Time: 1667566618.941394, Usage: 2.74%, Memory: 29576 kB
System Time: 1667566622.547050, Usage: 81.95%, Memory: 29576 kB
2022-11-04 13:57:31,885 [proc_stats.py:_watchdog_callback()] - EVENT LOOP BLOCKED: 30.31 seconds, total blocked count: 4
2022-11-04 13:57:32,089 [proc_stats.py:log_last_stats()] -
System Time: 1667566616.934498, Usage: 1.17%, Memory: 29576 kB
System Time: 1667566617.954353, Usage: 2.49%, Memory: 29576 kB
System Time: 1667566618.941394, Usage: 2.74%, Memory: 29576 kB
System Time: 1667566622.547050, Usage: 81.95%, Memory: 29576 kB
System Time: 1667566652.073158, Usage: 96.24%, Memory: 30744 kB
2022-11-04 13:58:01,002 [proc_stats.py:_watchdog_callback()] - EVENT LOOP BLOCKED: 29.12 seconds, total blocked count: 5
2022-11-04 13:58:01,203 [proc_stats.py:log_last_stats()] -
System Time: 1667566617.954353, Usage: 2.49%, Memory: 29576 kB
System Time: 1667566618.941394, Usage: 2.74%, Memory: 29576 kB
System Time: 1667566622.547050, Usage: 81.95%, Memory: 29576 kB
System Time: 1667566652.073158, Usage: 96.24%, Memory: 30744 kB
System Time: 1667566681.025304, Usage: 99.06%, Memory: 31456 kB
2022-11-04 13:58:40,060 [websockets.py:_process_messages()] - Websocket closed while writing: 1707607184
2022-11-04 13:58:44,191 [websockets.py:on_close()] - Websocket Closed: ID: 1707607184 Close Code: None, Close Reason: None, Pong Time Elapsed: 43.17
```

Hence I feel a statement that many functions need this is way oversimplified.  I could understand that a view on a certain limited set of files from startup can be useful, but such apparently unlimited fetching that can totally bog down the system with no clear indication on the cause just seems bad design to me.

meteyou, 2022-11-05:
Gcode search, themes autoloader and others just needs all directories in the filetree. How should mainsail just know if there is a theme, when it doesn't know all files?

Misterke, 2022-11-05:
I'll be addressing it another way.  Moonraker already has the concept of reserved paths, but unfortunately fails to stop traversal at those paths.  I'll try to submit a PR to Moonraker to fix that and then I have my own additional Moonraker component that allows me to add additional reserved paths to block Mainsail and others from drilling down into folders where it is not supposed to go.

Still I feel it does not make sense to drill through everything it can get its hands on.  Especially things like drilling down hidden folders (like .git is) doesn't make much sense and would in a search only cause bad results to be added.  But as said, I'll block it from the other end to avoid future performance problems on my setup.

meteyou, 2022-11-05:
We can also filter `.git` directories, but there has to be a special filter in added or changed directory/file notifications. Just filtering `.git` directories in the moonraker API would be a  good solution.

But... I would also prefer no git synced directories in the printer_data/config directory (or in other dirs that are "watching" by moonraker). The simplest solution would be to have the git synced directory in your home dir and link the config file (or a subdir) to your printer_data/config path.