mainsail-crew/mainsail #1151: Allow Support for Moonraker API Authentication

awickham10, 2022-11-14:
### Requested feature:

Exposing Moonraker to the Internet can be dangerous, but Moonraker supports the usage of API keys, which reduces that risk. If Mainsail could support authenticating to Moonraker with their API keys, we could (somewhat) safely expose Moonraker endpoints to the Internet, allowing for easier remote access configurations.

### Solves the following problem:

Reduces risk of exposing Moonraker to the Internet. Reduces complexity of remote access.

yang6991906, 2022-11-14:
Support your suggestion

meteyou, 2022-11-14:
And this function is NOT to make your installation save in the internet. No part of Klipper was build to be safe in the internet!

closed 2022-11-14 by meteyou

awickham10, 2022-11-15:
Thanks for the quick response. Is there a reason you think Moonraker's API authentication isn't sufficient enough? Definitely not an API security guru here but it seems like it's using pretty standard authentication mechanisms? It feels like a combination of CORS and this would make it somewhat safe aside from brute force attempts.

meteyou, 2022-11-15:
Because the developer said it...