mainsail-crew/mainsail #1154: Adjust individual or multiple mesh points

gwoodruff, 2022-11-14:
### Requested feature:

I would like a graphic representation of the probed mesh points with a way to select one or many individual points and then raise or lower them relative to the rest of the mesh.    

### Solves the following problem:

My bed is 400 x 400 and when the mesh is probed many points still need slight tweaking up or down compared to the rest. The original firmware had this feature and it was very useful.

meteyou, 2022-11-14:
this is not possible atm. There we need an API point to change points in Klipper itself. Mainsail can only use Klipper or Moonraker functions.

gwoodruff, 2022-11-14:
I tried posting in klipper but was told this would be a better place to ask (after the requisite inquiry on WHY I would need such a feature!

werkstrom, 2023-07-28:
This can be done _manually_ by editing the mesh points (the actual probed ones) in printer.cfg. (I know it's not what @gwoodruff  asked for, but it's how it can be done, if you really want/need to).

As we have the probed values, calculating modified values should be possible in Mainsail (i.e. letting the user add an offset value like -0.155 and subtracting that from the probed values returned by klipper/moonraker. From @th33xitus answer, it would not be possible to write to printer.cfg from the Mainail UI atm. It would however be possible to generate a text/string with the adjusted probe point values that the user can paste them selves into printer.cfg at the correct position.

For context I need to adjust the values since the probe on my Elegoo Neptune 3 MAX does not provide accurate readings at the edges (for any of a number of reasons) So whenever I probe I need to then print a test layer and adjust these values in order to get an absolute perfect first layer. I do it manually no but a UI would be awesome. (not asking anyone to do it, just pointing out a use case and a possible high level design idea)