mainsail-crew/mainsail #1156: Cameras set to UV4L-MJPEG no longer work after update to v2.4.0

HyperToxic33, 2022-11-15:
### Mainsail Version:

v2.4.0

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

Cameras set to UV4L-MJPEG no longer work after update to v2.4.0

### What did you expect to happen instead?

Cameras continue to work with a setting of UV4L-MJPEG

### How to reproduce this bug?

Change camera setting to UV4L-MJPEG

### Additional information:

[klippy.log](https://github.com/mainsail-crew/mainsail/files/10009758/klippy.log)
[moonraker.log](https://github.com/mainsail-crew/mainsail/files/10009759/moonraker.log)
[crowsnest.log](https://github.com/mainsail-crew/mainsail/files/10009760/crowsnest.log)
[image]

HyperToxic33, 2022-11-15:
Another user had the same problem after updating to v2.4.0, so I figured I'd report the issue.

cruiten, 2022-11-15:
I also encountered this exact problem after updating to Mainsail v2.4.0.

I reverted back to Mainsail v2.3.1 and my camera works again.

jalanjarosz, 2022-11-16:
Had same issue issue after update. 
Changed camera settings to either
MJPEG-Streamer or Adaptive MJPEG-Streamer and camera feed was restored. 

[image]

HyperToxic33, 2022-11-18:
I'm not sure what I'm supposed to do to get the camera to work again with UV4L-MJPEG.  I have the latest changes as of now and it still doesn't work.   What am I missing?

[image]
[image]

cruiten, 2022-11-19:
HyperToxic,

The fix for this problem was committed and merged into the Mainsail
development branch, it is not officially released yet.

To get this fix you will either have to make your own build of the Mainsail
development branch and deploy that to your printer's host computer, or you
will have to wait for the next official release...

ZoZo

PR #1158 (2022-11-16): Fix: bad printerUrl check in ternary operator
This fixes a problem introduced in v2.4.0 for webcams defined with service `Uv4l-Mjpeg`, and will fix #1156.