mainsail-crew/mainsail #1167: Dashboard sections can not be reordered

mtwhitley, 2022-11-25:
### Mainsail Version:

2.4.0

### Browser:

Chrome

### Device:

Other (Please tell us below)

### Operating System:

Other (Please tell us below)

### What happened?

Tested on both iOS (mobile) and MacOS (desktop). Dashboard sections can be moved from one column to another but can not be reordered vertically. 

### What did you expect to happen instead?

Sections (Webcam, Toolhead, Extruder, Miscellaneous, etc) should be adjusted based on user preference. They appear to reorder in the UI during the drag event, but when released the order is reverted to the previous state.

### How to reproduce this bug?

1. Visit the Mainsail interface
2. Click/Select the Settings/Gears icon in the top right
3. Browse to Interface Settings > Dashboard
4. Grab the reorder icon next to any given section
5. Drag to a different area and release
6. Observe the position has not changed

meteyou, 2022-11-25:
looks like a safari issue. cannot reproduce it in other browsers, only in safari.

mtwhitley, 2022-11-26:
It is visible for me in Mobile Chrome and Mobile Safari as well as Chrome on the Desktop.

screech24, 2022-11-26:
> It is visible for me in Mobile Chrome and Mobile Safari as well as Chrome on the Mac Desktop.

I’m noticing the same issues as well. Can confirm seems to be a Mac issue/iOS

meteyou, 2022-11-26:
> It is visible for me in Mobile Chrome and Mobile Safari as well as Chrome on the Mac Desktop.

All browsers on iOS devices have to use the Safari engine. So all iOS/iPad devices should have the same issue. I tested in on MacOS in Chrome and it worked fine. Only on MacOS Safari, i can reproduce this issue.

meteyou, 2022-11-27:
Since yesterday, I have the same issue also on chrome. So it looks like a generic issue...

Zodiak1993, 2022-11-27:
Same problem using the latest version of firefox.

meteyou, 2022-11-27:
I think I found the issue... working on a fix right now.

closed 2022-12-01 by meteyou

johyphenel, 2025-01-04:
This issue still exists in OrcaSlicer Device page on Mac — not sure what browser that uses under the hood. 

OrcaSlicer: 2.2.0
MacOS: 15.2

meteyou, 2025-01-04:
@johyphenel thx for your report! i tested it also with OrcaSlicer 2.2.0 on my mac and reproduce the issue. i also tested Safari, because i think orcaslicer use the OS webviewer in the background and in this case it should be Safari, but in safari it works fine for me.

Pls open a new issue (Bug Report) with Orcaslicer in the title, that we do not forget this issue. its just very hard to debug in orcaslicer.

johyphenel, 2025-01-04:
@meteyou filed https://github.com/mainsail-crew/mainsail/issues/2110

PR #1176 (2022-12-01): fix: fix dashboard interface settings
this PR fixed #1167