mainsail-crew/mainsail #1177: Overlay Print information on Webcam stream

SmallStrayFox, 2022-11-30:
### Requested feature:

During a print I spend most of my time on the Webcam tab so I can keep an eye on the print and any potential issues. It would be super handy if I could also see some live print stats overlaid on the webcam feed so I can keep an eye on important things like temperatures.

### Solves the following problem:

solves having to switch between the Dashboard and Webcam tabs, when it's much more preferable to have glanceable information on the webcam feed that allows us to see any given live variable for the current print (similar to the current fps counter).

### Additional information:

ideally should be able to include any variable that's being monitored on the dash in a format that is compact but readable. graphs are nice but honestly one line of text would be more readable, e.g. you'd edit a text string like

%temp_extruder% - %temp_bed% || %temp_mcu1% - %temp_pi% || %Percent_done% - %ETA%

and it showed overlaid on the live video feed as

Extruder: 210.0°C - Heater Bed: 60.0°C || Board MCU: 38.0°C - Raspberry Pi: 32.0°C || %85 - ETA: 17:35

apphead, 2022-12-19:
Would love to see that too!
@Turbine1991 Would you mind sharing your extensions?

Hotwire, 2023-07-12:
Any movement on this? It would be SUPER handy for the time lapse to have the info too.

meteyou, 2023-07-13:
@Hotwire this have nothing Todo with timelapse.

CoffeeSquirel, 2024-08-08:
Did anything come of this?

meteyou, 2024-08-08:
@CoffeeSquirel it's still open and no pr is linked to it. So it's just an open FR.

dfbpurcell, 2024-08-13:
How do we apply the patch file?

2wenty2wo, 2025-01-11:
Also very interested in this.