mainsail-crew/mainsail #1184: manual z window error loop

Avatarsia, 2022-12-06:
### Mainsail Version:

2,4

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

the bed level assist window wont disapper when you have missconfiguration in max positions
(max position 210, try to move to pos 250) 
when you press abort it says abort is unknowen. 
when you press x - unknowen
only way is to restart klipper via ssh 

### What did you expect to happen instead?

the x should close the window in any way

### How to reproduce this bug?

like written in the what happened part

meteyou, 2022-12-09:
the main problem is, that klipper doesnt reset the object. so the printer is stucking in this state. Pls create an issue in klipper.

closed 2022-12-09 by meteyou

Avatarsia, 2022-12-09:
no klipper is abort the state, but you cant close the z offset window. 
if the x would close the window and give the interface free instead send the command abort, all should be fine

meteyou, 2022-12-09:
That's not right. If it would stop completely, the window would close.

meteyou, 2022-12-10:
but this would not fix your issue. every time you open your webinterface again, it would open this window again, because klipper is in this state and send the state to open this window. thats why i wrote -> pls report this to klipper itself!

meteyou, 2022-12-10:
https://www.klipper3d.org/Status_Reference.html#manual_probe

the value `printer.manual_probe.is_active` don't switch to `false`. this is the real issue. i can just close the window in the gui, but it will pop up every time you refresh the browser window or open the tab again until you restart klipper, or klipper fix this issue itself.