mainsail-crew/mainsail #1186: Map Macros  / Manual Jog to Keyboard

GlennoFromOZ, 2022-12-09:
### Requested feature:

Would there be any possibility to map macros or other controls (Like manual jog), a keyboard ?  

I see that there seems to be the possibility to map dedicated hardware buttons connected to the pi via klipper somehow, but having something easyt to setup via mainsail would be great.

RREE, 2023-04-24:
This could be similar to what UGS (Universal G-Code Sender) provides. See for example this [video](https://www.youtube.com/watch?v=RefxgL2kG2Q)

Aitch-ko, 2026-11-01:
For anyone who lands here for the jog part: I built [jogpad](https://github.com/Aitch-ko/jogpad), a small service that runs next to Klipper. It turns a USB keyboard, numpad or macro pad with knobs, plugged into the Klipper computer, into a jog pendant.

- One press or knob detent is one step: 0.01, 0.1 or 1 mm, switched with a key.
- A held key keeps the axis moving until you let go.
- Every move shows up in the Mainsail console, and a web page sets up the keys.

It works without looking:

- you feel the keys and the knob detents
- the pad lights up in the colour of the step size
- the printer's beeper counts the step size, and a long beep tells you a key was refused

Similar to the UGS pendant mentioned above, but it talks to Klipper through Moonraker. So it is not the Mainsail feature asked for here, and it does not run macros.

If you have low vision and give it a try, I would like to hear what is still missing: https://github.com/Aitch-ko/jogpad/discussions/24

Docs: https://aitch-ko.github.io/jogpad/

meteyou, 2026-11-02:
@Aitch-ko  This is not the place for "self promotion". Feel free to use the discord server for that. The GitHub issue tracker is the todo list (backlog) for Mainsail.