mainsail-crew/mainsail #1199: Temperature Graph gets stuck

RoadRunnr, 2022-12-18:
### Mainsail Version:

2.4.1

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Linux

### What happened?

Temperature graph stops updating. Chrome dev console has this:
```
index.244641ff.js:191 Uncaught TypeError: Cannot read properties of null (reading 'clone')
    at isTwoLabelOverlapped (index.244641ff.js:191:102495)
    at fixMinMaxLabelShow (index.244641ff.js:191:102310)
    at Object.axisTickLabel (index.244641ff.js:191:100015)
    at o.add (index.244641ff.js:191:98062)
    at Array.forEach (<anonymous>)
    at each$4 (index.244641ff.js:167:5836)
    at l.render (index.244641ff.js:191:111682)
    at index.244641ff.js:189:71925
    at Array.forEach (<anonymous>)
    at each$4 (index.244641ff.js:167:5836)
```

### What did you expect to happen instead?

Graph keep updating

### How to reproduce this bug?

Run a print.

### Additional information:

Chrome 110.0.5478.4 (Official Build) dev (64-bit)

meteyou, 2022-12-18:
sorry, but only with this snipped and without any logs its not possible to reproduce this issue.

closed 2022-12-18 by meteyou

RoadRunnr, 2022-12-19:
Closing this issue without even mentioning what log files you need is not very helpful. There are no bug reporting guidelines for this project that give any hints which log files you would need.

moonraker.log, klippy.log and nginx access logs contain nothing helpful.

Here is the full console log of the problems:
[mainsail-1671455565933.log.gz](https://github.com/mainsail-crew/mainsail/files/10259494/mainsail-1671455565933.log.gz)