mainsail-crew/mainsail #1206: Extruder select drop-down list has no effect

naikymen, 2022-12-22:
### Mainsail Version:

2.4.1

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Linux

### What happened?

There is a dropdown list to select an extruder, on the "Extruder" panel.

[image]

I selected the second extruder (named "extruder1").

No command was sent.

Then I pressed the button to move the extruder. The second extruder should have moved.

However, the first extruder (named "extruder") moved instead.

### What did you expect to happen instead?

I expected that after selecting an extruder from the drop-down list, it would be selected as the active extruder, and pressing the move extruder buttons would move the selected extruder.

I expected that selecting an extruder from the list would send an appropriate "ACTIVATE_EXTRUDER" command to Klipper.

### How to reproduce this bug?

Configure two extruders.

### Additional information:

By searching in the mainsail-crew GitHub org, I found a config file showing T0 and T1 macros which do have the ACTIVATE_EXTRUDER command.

Search: https://github.com/search?q=org%3Amainsail-crew+ACTIVATE_EXTRUDER&type=code

By adding those to my config, new buttons appear on the Extruder panel at the Dashboard:

[image]

Additionally, pressing on T0/T1 updates the value on the drop-down list, but changing the value on the drop-down list does not change the tool.

The behaviour of the drop-down list is confusing, and the need for tool-changing macros named T0 and T1 is not documented (afaik).

Thanks for the great software!

naikymen, 2022-12-22:
Tagging #88

naikymen, 2022-12-22:
I see two alternatives:

- Disable the drop-down list (gray it out), and maybe add a help/hint link to the documentation about T0/T1 macros.
- Have the drop-down list send an ACTIVATE_EXTRUDER command on change, or use the T0/T1 macros if available.

meteyou, 2022-12-22:
The drop down is only for pressure advanced. Which extruder should be changed

naikymen, 2022-12-22:
Ok, good to know.

I still think that the behaviour is confusing, so I'll close this and make a feature request.

Thanks!

closed 2022-12-22 by naikymen