mainsail-crew/mainsail #1213: Page refresh needed when disabling Voxel mode in G-Code Viewer

markwalet, 2022-12-29:
### Mainsail Version:

v2.4.1

### Browser:

Chrome

### Device:

Laptop

### Operating System:

macOS

### What happened?

When disabling Voxel Mode, a re-render triggers, but no changes are visible.

### What did you expect to happen instead?

The model should be automatically re-rendered without the need to refresh the page.

### How to reproduce this bug?

- Go to G-Code Viewer page and have Voxel Mode enabled (refresh the page just to be sure
- Enable Voxel Mode
- Refresh page
- Select model to render
- Disable Voxel Mode

### Additional information:

It's only happening when disabling Voxel Mode. Enabling Voxel Mode is always directly shown.

I also found an error in the console, but I'm not sure if it has anything to do with the bug. It also happens when enabling Voxel Mode, which :

```
RangeError: Maximum call stack size exceeded
    at a.<anonymous> (Viewer.5c7332ad.js:14232:9920)
    at h (Viewer.5c7332ad.js:14229:5825)
    at Generator._invoke (Viewer.5c7332ad.js:14229:5567)
    at Generator.next (Viewer.5c7332ad.js:14229:6184)
    at Lp (Viewer.5c7332ad.js:14229:11145)
    at n (Viewer.5c7332ad.js:14229:11342)
    at Viewer.5c7332ad.js:14229:11403
    at new Promise (<anonymous>)
    at a.<anonymous> (Viewer.5c7332ad.js:14229:11289)
    at ts.value (Viewer.5c7332ad.js:14232:10082)
```

I'm also not sure what Voxel Model should do anyway. I'm getting a bunch of lines which are not very useful to me, a quick Google Search shows me all kinds of different examples. So maybe the rendering method is broken alltogether. Hard to say something useful about that when you don't know what Voxel rendering is ;p I do know that the changing rendering method should auto update the result, that's why I created this ticket.

Result:

[image]

Sindarius, 2023-01-06:
Voxel mode is not working properly at the moment. Essentially is renders the model similar to what you would see in Minecraft. This mode was mainly designed for visualizing ASMBL where blocks can be added by printing and removed by milling in a single machine (toolchanger).

Sindarius, 2023-01-08:
I have found the issue in the Viewer package that is causing this to fail.

Also here is an example of what you'd see in voxel mode.
Voxel
[image]
Normal
[image]

markwalet, 2023-01-10:
Ah alright, I'm not sure if there's already a general Voxel mode bug issue. If not, we can make this the general issue to keep track of progress. If there is, we can just close this one and fix it there

Sindarius, 2023-01-10:
It is already fixed in the viewer library and should be in the next Mainsail release.

closed 2023-01-10 by markwalet