mainsail-crew/mainsail #1225: Portuguese Brasil Translation of Mainsail

adrianobarrosmba, 2023-01-12:
### Requested feature:

I have successfully translated Mainsail into Portuguese Brazil
,can you please add the traslation as soon as possible? (.json file is in the attachment)

[pt_br.zip](https://github.com/mainsail-crew/mainsail/files/10402817/pt_br.zip)

meteyou, 2023-01-20:
Hey! thx for the translation. I checked your file, but there are many empty strings. Please fill in all translations, and then I can upload it.

th33xitus, 2023-02-10:
@adrianobarrosmba if you want the translation to be integrated into mainsail, please provide a complete translation. no empty strings allowed. filling in half of the file with empty strings makes the whole translation rather useless and will actually do more damage than it helps, since all those empty strings will result in missing text in the application.

adrianobarrosmba, 2023-02-10:
ok, i'll be completing the translation, sorry i didn't know you had empty items.

closed 2023-03-04 by github-actions[bot]