mainsail-crew/mainsail #1229: Mainsail making chrome extremely slow and break other website rendering (youtube especially)

akhamar, 2023-01-15:
### Mainsail Version:

2.4.1

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

Since the last few update (since 2.4.1 ?) I had strong slow down on other web page (especially Youtube video playing).
Having mainsail open (background or foreground in tabs) make the whole chrome to be slow and having problem rendering other website.

### What did you expect to happen instead?

Not having other website in separate tabs having rendering issues.

### How to reproduce this bug?

Reproduce is extremely easy.
Open youtube and watch a video (4K), see that video is rendering correctly. Now open mainsail in a separate tab, and see that youtube start stuttering, video feed break, etc.

akhamar, 2023-01-15:
I'll add that this problem only appear when on the dashboard of mainsail.
[image]

akhamar, 2023-01-15:
I've narrowed down on the dashboard what's causing the issue :
- temperatures
- miscellaneous

Closing down those two panel make the problem goes "away".

[image]

meteyou, 2023-01-15:
the tempchart needs a lot of performace. this is not a bug. its just needs this performance to display it. feel free to disable the tempchart if you dont like it.

akhamar, 2023-01-15:
I wasn't getting this problem from previous version of mainsail.
And if just a single page break the whole browser it's definitely a bug.

meteyou, 2023-01-15:
the tempchart is only on the dashboard. if you open another page it will unmount this component and it dont need any performace. feel free to check the changelog. there is no difference in this area in the newer releases.

akhamar, 2023-01-15:
After further research it's the moving icons posing problems

[image]

meteyou, 2023-01-15:
i think these icons moving since v0.3.0...

akhamar, 2023-01-15:
Maybe a chrome update made them behave incorrectly. Or the way of making them move since an old version wasn't really the way to do it. Dunno.

meteyou, 2023-01-15:
I cannot reproduce an issue with a youtube 4k video, but I see a difference in the chrome task manager.

akhamar, 2023-01-15:
I think it's also due to chrome bringing some bug or change. I've seen that other website with animation (rotation and such) have the same kind of behavior.
Might be a good idea to look at chrome issue regarding this situation.

Also rather than just removing the animation, make it so we can chose between animated or not maybe ?

meteyou, 2023-01-15:
It would also be possible, but I don't see any benefit for this little "function"/"Gucci stuff".

I will double check it in other browsers to be sure it is only a chrome bug or a general performance issue.

akhamar, 2023-01-16:
As you said the rotating animation was already present since a "long time ago". I would concur that a recent chrome update somehow killed the perf of this particular kind of html/css transformation.

th33xitus, 2023-01-29:
im unable to reproduce the issue on chrome 108.0.5359.125. i had 3 spinning fan icons on the dashboard AND the temperature chart, while watching this video in 4k: https://www.youtube.com/watch?v=z24WRrAmdwc

I don't even see a high cpu load in the chrome task manager while mainsail running in the background (on the dashboard) and the video running in the foreground. no stuttering of the video at all and the cpu load of the mainsail tab rarely just spikes from 0% to 1.5%.

i see that your chrome version is exactly 1 off compared to mine. i will update my chrome and test again.

closed 2023-03-12 by meteyou

LimesKey, 2023-04-05:
I am receiving this issue too, both with Chrome and FireFox, more prominently with Firefox. It seems to be caused by the webcam and somewhat by the temperature tab. With the webcam tab open, the tab consumes 2GB of RAM and 100% of my CPU on my laptop. The webcam is 1080p 30FPS. The webcam tab also tends to be very slow to refresh, even though it reports to be 30fps, the image is only being refreshed every few seconds. As you can see in the photos it shows the resources the tab consumes and that the webcam tab is greyed out as it usually takes 10 - 20 seconds to load.

[image]
[image]

PR #1232 (2023-03-12): feat: allow fan animations to be disabled to save browser perf.
This PR will save performance in the browser.

fixes #1229.