mainsail-crew/mainsail #1234: How to add UI control for z1, z2, and z3 stepper motors

neon5d, 2023-01-17:
### Requested feature:

Hello, 
I wonder how I can add UI control for the z1, z2, and z3 stepper motors.
Can you explain the procedure or the documentation?
Thank you

meteyou, 2023-01-17:
All z-steppers will combine into one axis. Therefore you can only move them all together via the GUI.

But there is also a force_move command in Klipper. This command moves steppers and not axes. You can address single steppers with this command (same as the buzz_stepper command).

But we don't use GitHub for support. Only for bugs or feature requests. So please use our discord for such questions.

closed 2023-01-17 by github-actions[bot]

neon5d, 2023-01-17:
Thank you for the information.
I will check the Discord-Server to get some more information.
Thank you.