mainsail-crew/mainsail #1235: Unable to see highlighted text

BlackNet, 2023-01-17:
### Mainsail Version:

v2.4.1

### Browser:

Other (Please tell us below)

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

used: chrome, edge, firefox, safari. 

when a line is selected, and a word is highlighted, I am unable to see the cursor or the highlighted section because it lacks contrast.  This comes into play when you cut/copy/paste.

This perhaps is a colorblind problem, but with its current state, I am unable to change anything dealing with the editor to make it better visible for me.,

### What did you expect to happen instead?

To see what I was highlighting and where the cursor is at.

### How to reproduce this bug?

1) Be colorblind

2) Highlight any word

3) Try to find the word you just highlighted because it lacks contrast.

meteyou, 2023-01-17:
Thx for your bug report! Can I post screenshots to verify if the colour matches better for colourblind users?

BlackNet, 2023-01-17:
[image]

The attached photo is one example.  I have some letters highlighted; where is my cursor at?  If you zoom in really well you can see it, but as it is by default, you can not.  Going to a small resolution may fix it but that would make the edit mostly unusable

> Thx for your bug report! Can I post screenshots to verify if the colour fits better for colourblind users?

BlackNet, 2023-01-17:
> Thx for your bug report! Can I post screenshots to verify if the colour fits better for colourblind users?

Here is a zoomed in image of the problem. You can see the highlight and cursor at this level.  It is to subtle and not enough contrast for normal viewing even if you are not colorblind, we just have a harder time with these areas. 

 [image]

TxBillbr, 2023-02-08:
This is not just a color-blindness problem. It is a problem for all who use the dark theme (especially on large screens)

BlackNet, 2023-02-22:
would seem this issue has been fixed.  Have confirmed it with multiple people across several setups.

closed 2023-02-22 by BlackNet