mainsail-crew/mainsail #1249: Typing line invisible - Text Editor

4e576rt8uh9ij9okp, 2023-01-29:
### Mainsail Version:

2.4.1

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Linux

### What happened?

Hello there,
So when editing the printer.cfg file (and any other file) when the text editor opens up and you move the `|` "typing line".
It feels like there is a blinking delay or it's invisible for some reason.

You open the printer.file (or any other) and you see the `|` at the start blinking, now when you press an arrow on your keyboard the line moves but it's invisible. Like it would be "display:none" when it's being moved or there is a delay or something.
Because usually when you move the line in editors it is VISIBLE and not blinking, after you stop moving it with arrows or "mouse clicking" it starts blinking again.

### What did you expect to happen instead?

Read "what happen"

### How to reproduce this bug?

Read "what happen"

th33xitus, 2023-01-30:
Im not fully understanding your description but i at least spotted indeed something "odd" with the cursor going on. Is it this phenomena you tried to describe? When the blinking frequency of the cursor seems to change and it seems that there are kinda two different cursors?
https://user-images.githubusercontent.com/31533186/215519243-ec251f25-14a8-4145-9119-8464335948a9.mp4

BlackNet, 2023-02-03:
This is the same issue as [https://github.com/mainsail-crew/mainsail/issues/1235](url)

closed 2023-02-24 by github-actions[bot]