mainsail-crew/mainsail #1251: Percentage complete does not update if tab is unfocused

nonlinear42, 2023-01-31:
### Mainsail Version:

v2.4.1 

### Browser:

Firefox

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

The percentage complete displayed in the tab of a print job does not update but stays on the last updated percentage using Firefox 109.0 (64-bit) on Windows 10 when the tab with Mainsail interface does not have the focus. Interestingly, the "pie chart" style visualization icon seems to always update, but not the text with the percentage beside it. As soon as the tab has the focus, the text is updated to reflect the correct percentage. If the tab is left defocused when the print job finishes, the text in the tab seems to always update to "Complete" and the pie-chart visualization disappears even without having focus.

This issue seems to only occur when the Mainsail interface is left on the main dashboard, but seems to never occur when switching to the gcode viewer or the Machine section in the interface. Other sections of the Mainsail interface have not been tested by me for this issue.

### What did you expect to happen instead?

Tab always updates even without focus.

### How to reproduce this bug?

Start a print job, make sure Mainsail is showing the main dashboard, then switch to another browser tab. The text indicating the percentage complete of the print job will not update, but the pie-chart icon will.

3dcase, 2024-12-10:
Same on Linux 24.04.1 LTS with firefox.

3dcase, 2024-12-11:
Same on windows with Firefox BUT windows with Edge or Chrome update correctly.

meteyou, 2024-12-11:
@3dcase thx for your feedback. normally i just use Chrome or Arc on MacOS and also so no issues here. so it looks like a firefox specific issue.

3dcase, 2024-12-18:
I asked Perplexity about the firefox feature of browser.tab.unload

what is the effect of setting browser.tabs.unloadOnLowMemory to false in firefox? Are there any negative consequences?
Sources
Answer
Setting browser.tabs.unloadOnLowMemory to false in Firefox disables the automatic unloading of inactive tabs when the system is low on memory. This feature was introduced to help manage memory usage and reduce crashes in low-memory situations by unloading tabs that are not actively being used. Here are the effects and potential negative consequences of this setting:
Effects of Disabling Tab Unloading

    Increased Memory Usage: By disabling this feature, Firefox will retain all open tabs in memory, which can lead to higher overall memory consumption. This is particularly significant for users with limited RAM, as it may result in slower performance or system instability if memory runs low[2](https://support.mozilla.org/ur/kb/unload-inactive-tabs-save-system-memory-firefox)
    [3](https://www.ghacks.net/2021/09/10/firefox-will-soon-unload-tabs-to-cope-with-low-memory-and-reduce-crashes/)
    .
    Reduced Crashes: While disabling tab unloading may prevent some tabs from being unloaded unexpectedly, it can also increase the likelihood of crashes if the browser exceeds available memory. The unloading feature was designed to mitigate out-of-memory crashes by freeing up resources[4](https://firefox-source-docs.mozilla.org/browser/tabunloader/)
    [5](https://support.mozilla.org/en-US/kb/unload-inactive-tabs-save-system-memory-firefox)
    .
    User Control: Users may prefer to manage their tabs manually, allowing them to keep all tabs open without interruption. However, this comes at the cost of potentially overwhelming system resources during heavy browsing sessions[6](https://www.reddit.com/r/firefox/comments/co6xng/browsertabsunloadonlowmemory_how_much_is_low/)
    [7](https://superuser.com/questions/1717123/disable-firefoxs-tab-unloading-feature)
    .

Potential Negative Consequences

    Performance Issues: On systems with limited RAM (e.g., 8 GB or less), disabling this feature can lead to performance degradation as Firefox may struggle to handle multiple active tabs simultaneously. Users may experience lag or slow responsiveness during browsing sessions[3](https://www.ghacks.net/2021/09/10/firefox-will-soon-unload-tabs-to-cope-with-low-memory-and-reduce-crashes/)
    [4](https://firefox-source-docs.mozilla.org/browser/tabunloader/)
    .
    Increased Risk of Crashes: Without the ability to unload tabs, users may encounter more frequent crashes during high-memory usage scenarios since Firefox will not have the option to free up resources automatically[4](https://firefox-source-docs.mozilla.org/browser/tabunloader/)
    [5](https://support.mozilla.org/en-US/kb/unload-inactive-tabs-save-system-memory-firefox)
    .
    Impact on System Resources: Keeping many tabs open can significantly consume CPU and RAM, which might affect other applications running on the system, leading to a less optimal overall computing experience[7](https://superuser.com/questions/1717123/disable-firefoxs-tab-unloading-feature)
    [8](https://superuser.com/questions/1740578/how-to-reduce-the-firefox-memory-usage-as-much-as-possible)
    .

In summary, while setting browser.tabs.unloadOnLowMemory to false allows for a more stable tab experience without unexpected unloading, it can lead to increased memory usage and a higher risk of crashes on systems with limited resources. Users should weigh these factors based on their specific hardware capabilities and browsing habits before making this change.

Maybe this is helping but does not seem a good idea to switch that off.

meteyou, 2024-12-20:
> Maybe this is helping but does not seem a good idea to switch that off.

To be honest... this is the default setting for each other browser... so i dont think it is a "really bad idea"...

3dcase, 2024-12-20:
> > Maybe this is helping but does not seem a good idea to switch that off.
> 
> To be honest... this is the default setting for each other browser... so i dont think it is a "really bad idea"...

I am no expert in browsers and I have no idea how different FF works under the bonnet from those other browsers. But they must have had their reasons and they do not state how many open tabs is too much so I choose to stay safe and live with that tab not updating, its not critical, just would be nice.